njfaria/SolPipLog #2: comunication USB HID

lzahradil, 2017-01-09:
Hello,
I switched inverter PIP MKS for MSXE. MKS had a variant RS-232 and I had no problem with communication.
MSXE variant has a USB HID and the script can not read data as RS-232 variants.
I need to ask the inverter to commands QPIRI and save the result to a file.
Can you please help communicate how you solved? In your solarpiplog runs fine

njfaria, 2017-01-31:
You need to comunicate with the Inverter as a HID device. i am using Libusb as library. You got to read the USB specs. A little complicated.

lzahradil, 2017-01-31:
I created a script that communicates via libusb. The data is sent successfully. Answer pull out from buffer. I have a problem when you do not come from the buffer all the data and script sends a new command. Followed by NAK.
Is it possible to publish the source code that provides data read? Or how do I know that the data is correctly received and can send a request for further reading?

Many thanks

closed 2017-12-10 by njfaria