njfaria/SolPipLog #3: Raspberry Pi 3 getting error on EMONCMS: server emoncms.org not responding

salman373, 2017-04-23:
Send to PVOut is working fine but getting error on send to EmonCMS. I am using my correct write key that was working fine on ICC and AICC.
Plz help me to resolve this issue. Thanks.

[image]

salman373, 2017-04-25:
I have Wait one full day for EmonCms but getting the same error. Plz help me to resolve this issue.

[image]

salman373, 2017-04-30:
Can any one help me, my problem still not resolved. Solpiplog can not pushing data on emoncms.
geting error "server emoncms.org not responding."

plz help me. Thanks.

njfaria, 2017-05-09:
how "bad" is your internet connection ?
try ping to emoncms.org. how many fails do you have ?

salman373, 2017-05-09:
My internet is very good. please check the ping result mention below.

[image: Inline image 1]

salman373, 2017-05-09:
[image]

salman373, 2017-06-02:
Issue has been resolved, it is working. Thanks for the support.

closed 2017-06-02 by salman373