njfaria/SolPipLog #6: AutoStart

salman373, 2017-11-26:
i want to auto start the Solpiplog with saved configurations. is it possible to autorun this app when pi restart on any error e.g low voltage or any other issue ?

dourix, 2017-11-28:
will be great indeed...

njfaria, 2017-12-06:
i am going to publish a new version with autostart. i will publish a link with the image, and a readme how to install autostart for the ones who dont care about images.
github does not alow big files, or is there any other solution ???

salman373, 2017-12-08:
Thank you very much.
Please also add error logs. e.g. low voltage, low battery etc. error and error date and time.
Thanks.
Did you publish the autostart link as i did not find this on github.
Thanks.

dourix, 2018-04-14:
looks working fine; many thanks !

closed 2018-06-03 by njfaria