njfaria/SolPipLog #7: Idea: Command line only version

mdhemmi, 2018-01-17:
Hello,

would be possible to create a command line only version where you can define the output in a config file. For example output to Emoncms, influxdb or even a text file.

Thanks
hemmi

abc2006, 2019-02-18:
Yeah, i miss this right now!
Would support you for that feature...

njfaria, 2019-04-06:
new version is sending some values over MQTT.
more flexible than sending to a text file.

mdhemmi, 2019-06-19:
Thanks for MQTT much appreciated. Would be great if your application could run on an RPI without desktop as a service.

GeirTS, 2019-08-01:
Is it possible to add all of the inverterdata (qpigs) as MQTT-values

njfaria, 2019-08-02:
sending more values as MQTT on version v2.5.5.20190802

closed 2019-08-02 by njfaria

kxn25, 2019-11-18:
When is the Releas planed?

jmanley, 2020-02-08:
I realise this ticket's been closed, but I don't see a cli option in the current version.  This feature would be very handy for me, too.  Is the conclusion that this feature will not be implemented?