njfaria/SolPipLog #10: Doesn't work on Raspberry Pi Zero W

maurangzaib, 2018-04-10:
Encountered following error while running ./solpiplogstart.sh.

An unhandled exception occurred at $0002F4E4:
EAccessViolation: Access violation
  $0002F4E4

Working fine on RPi 3

njfaria, 2018-08-02:
pi zero w version on solpiplog.tk

closed 2018-08-02 by njfaria

njfaria, 2018-08-02:
When did you downloaded the pi zero version ? A first one did got a bug. I think the version online is working.

maurangzaib, 2018-08-02:
I downloaded in last week. Just renamed existing solpiplog executable file
in sol folder and placed executable of pi zero version, but when run same
exception reported.

Please guide if I am missing any step.

Regards,

maurangzaib, 2018-08-02:
Hi,

Today again I downloaded and tried to run the executable but same error. I
am attaching screenshot of error.

Please guide if I am missing any step.

Regards,

maurangzaib, 2018-08-03:
Kindly update. I am desparate to run the software on Rpi zero.

Will appreciate your kind support.

Regards,

njfaria, 2018-08-03:
HI. File was corrupted. Link is fixed. try again. Sorry for delay

maurangzaib, 2018-08-04:
> Hi,
>
> Thanks for your reply.
>
> I tried the executable file mentioned in the link earlier on RPi zero W,
> but it is not working and throws following error:
>
> "Segmentation Fault"
>
> I installed the code as per instructions and replaced only the "solpiplog"
> executable for RPi zero but not working.
>
> Kindly guide.
>
> Regards,

Hi,

It run fine and continued to upload to emoncms and pvoutput to around 1
hour. Then after program terminated with same "Segmentation Fault" error. I
executed several time but it aborts randomly giving same message.

Screen shot is attached. Kindly guide what could be the issue.

Regards,

maurangzaib, 2018-08-04:
Also following error is reported intermittently in Debug Log window and
need to press Start button again. Check box on option "connect to inverter
on" need to click again:

04-08-2018 19:33:39 no device found-4

Please guide why the disconnection occurs intermittently by giving above
error.

Regards,

maurangzaib, 2018-08-13:
Hi,

Kindly update as application is crashing intermittently on pi zero w by
giving "no device found-4" message. When run again it works for some time
then again gives same message. Btw I checked same on rPI 3 and it works
fine there.

Kindly guide.

Regards,