njfaria/SolPipLog #12: emoncms posting doesnt work

solarsnoop, 2018-06-29:
I setup the emoncms logging to my own emoncms server. But SolPipLog has an error message:

Internet/HTTP Error: 406 
when talking to: https:/....zapto.org/emoncms/input/post.json?node=4&json=%257B1:001.0,2:00.0,3:228.0,4:50.0,5:0000,6:0000,7:000,8:371,9:25.69,10:000,11:067,12:0470,13:0000,14:025.7,15:25.82,16:00001,17:00000,18:0,19:1,20:0,30:2018,31:06,32:29,33:21,34:23,35:18,36:0,37:0,38:10010110,39:000%257D&apikey=dd......

So first is the API for EMONCMS should be:
https:/......zapto.org/input/post?node so when i copy this link and remove "/emoncms/"

https:/....zapto.org/input/post.json?node=4&json=%257B1:001.0,2:00.0,3:228.0,4:50.0,5:0000,6:0000,7:000,8:371,9:25.69,10:000,11:067,12:0470,13:0000,14:025.7,15:25.82,16:00001,17:00000,18:0,19:1,20:0,30:2018,31:06,32:29,33:21,34:23,35:18,36:0,37:0,38:10010110,39:000%257D&apikey=dd......

I get the error: {"success": false, "message": "Format error, json value is not numeric"}

Why the emon post link has an /emoncms/ (this i could solve by change the appache conf.)?
But the form seems not correct. 

What i am doing wrong?

solarsnoop, 2018-07-04:
I found out that it must have to do with https redirection on my server. SolPipLog post it in http port 80. If I post it to my local PI where i setup only http, it works fine.

njfaria, 2018-08-02:
ok fine.

closed 2018-08-02 by njfaria