njfaria/SolPipLog #20: Time Switch Setting

salman373, 2019-04-30:
Hi,
I am very happy to get the Time Switch setting and it is working fine. but a small issue with me that is when it switch to Solar then i get SBU setting that is fine. but when it convert back to Grid it shift to UTI but i want it to be in SOL setting when back to grid.
Each day end i convert it to SOL and in Morning it switch fine to SBU but in evening it should convert back to SOL not to UTI.
Thanks

njfaria, 2019-05-03:
If you set SOL it changes to Grid at evening ( no Sun )
SOL is Solar first then battery. Grid provides power only if Solar is not available
 As long as the sun is shining you are on Solar/Battery
After Sun is gonne it changes to Grid.
You wouldnt either need the timeswitch....
but i should add switch to SOL

njfaria, 2019-07-26:
v2.5.5 suporting switch to SOL

closed 2019-07-26 by njfaria