njfaria/SolPipLog #25: SolPipLog v2.3.0 (Gandalf) Hangs

Schmidt-Heinrich, 2020-01-14:
This is an intermittent problem that I am struggling to find the source of.

Using a Pi4-4GB model.

I wrote a "emonpi 17Oct19" image as a base.
Loaded all the required GUI elements for Raspbian
Next I loaded SolPipLog v2.3.0

My suspicion is that I have a buffer issue of some kind, but I am grasping here.

Any advice or solution I will be grateful for.

GREAT product!

steverrobertson, 2020-01-26:
I had a similar problem during my installation. I had a 3 amp power supply that was barely producing 1.5 amps. It worked reliably until I plugged it into the inverter.  My thinking is the logic in the PIP uses the power from the USB on the raspberry to function. It always worked fine until I plugged it in, and then it would hang at very random times.

Schmidt-Heinrich, 2020-01-26:
Thank you for sharing you experience. 
I have tried two different Raspberry supplied power supplies, same result. 

I have scheduled a reboot at midnight for now. Of course this is not a solution but will remove the impact at this stage.

njfaria, 2020-02-16:
I presume you are not using emoncms. what errors are you having ?
I could make a image available with all configured.
How do you start solpiplog ?
execute "top" or "htop" from commandline and you will see processes that are running and with high CPU usage.
I have a PI4 and i am using the official power adapter from them.
But i have to say that i am using solpiplog at this moment on a Raspberry 3b+
The PI4 is for programing.

closed 2020-03-19 by njfaria