njfaria/SolPipLog #26: Pylontech Support

cybertza, 2020-01-21:
Hi do you have any plans to support Pylontech?

I am able to use it via serial with solar-sis but like your frontend.

https://github.com/cybertza/solar-sis/blob/master/example/US3000+US2000/calls.json

njfaria, 2020-02-16:
i dont have a pylontech, need to see the protokoll.

cybertza, 2020-02-17:
correct, thats why i linked the protocol in the calls.json link in the post.

cybertza, 2020-02-17:
but alternatively if you support battery data via mqtt, then it would also help a lot, for the meantime i will start migrating to your platform when i have a moment.

njfaria, 2020-02-20:
looking at the protokol, i see the guys who made that, are in the forum where i am posting too.

njfaria, 2020-03-19:
solved with your help

closed 2020-03-19 by njfaria

dc3tdriver, 2020-04-18:
Was there a final outcome to the pylontech support query? Is there a way to communicate with pylontech through a serial to USB converter cable?

njfaria, 2020-04-18:
i got one user which is using it, dont know how he connected it.
i do not have one, i could post you a link with a activated pylon version
if you are interested to try.

à(s) 13:39:

cybertza, 2020-04-19:
USB => console port @dc3tdriver - it is 232. on the RJ12 port, pinouts are in the manual.

overdrive2max, 2020-04-22:
> i got one user which is using it, dont know how he connected it. i do not have one, i could post you a link with a activated pylon version if you are interested to try. dc3tdriver <notifications@github.com> escreveu no dia sábado, 18/04/2020 à(s) 13:39:
> […](#)

Hi
Can I also try out the version with activated Pylon?

dc3tdriver, 2020-04-27:
Thank you Njfaria, yes  please send me the link. I am at home now so can give it a try.