njfaria/SolPipLog #32: Vm Ubnt 18.04

cybertza, 2020-02-19:
Hi I am getting the following error, Any advice, or what platform needs to be avail?

[image]

bash: ./solpiplog: cannot execute binary file: Exec format Error

cybertza, 2020-02-19:
i installed Debian, thinking that it may have been the issue, it was not...
[image]

cybertza, 2020-02-20:
So, i think i have tracked the problem down to the that the bin is only for ARM?

Is it possible to compile for amd64 as well?
Or use something more industry standard like docker to allow cross platform?

njfaria, 2020-02-20:
yes its based on arm. I will compile a version for amd64. I did not played to much around with docker yet.

cybertza, 2020-02-20:
ok thank you, looking forward to testing it then, should i load ubuntu 19.10 so long?
i can also create access for you if need be to test the pylon stuff if you are interested?

njfaria, 2020-02-28:
feel free to try it out...
https://my.pcloud.com/publink/show?code=XZbLADkZXWVDTGOBU6XqEXKW5ykQmBfVix8k
do you got access open, for the pylon stuff ?

cybertza, 2020-03-02:
Hey, Have sent you an email re Remote access for pylon stuff

cybertza, 2020-03-02:
Thank you, the build works.

Found some minor issues :p

closed 2020-03-19 by njfaria

uploader18, 2020-06-02:
would you compile it fir i386 debian? thanks