njfaria/SolPipLog #33: No longer able to get data from inverter

Bilalaerox, 2020-02-25:
initially i was succeeded to get get data from my inverter. but once i restarted my PI, i am no longer getting data through pvouput.org, although the inverter connects to PI through usb but it doesn't retrieve the data.
some snaps are for your kind consideration
[image]
[image]
[image]
 I will be thankful if someone points out what iam doing wrong.

njfaria, 2020-02-28:
do you have the hybrid version ?
if not and you didnt updated the software you need to tick "single" on inverter type

Bilalaerox, 2020-02-29:
> do you have the hybrid version ?
> if not and you didnt updated the software you need to tick "single" on inverter type

Thanks working now, But I cannot find value for PV input watts under inputs tab, Please help me to identify it.

njfaria, 2020-03-02:
just make me a print screen of the hybrid tab

closed 2020-03-28 by Bilalaerox