njfaria/SolPipLog #35: MQTT not starting correctly on reboot with 2.4.0 - maybe implement auto-reconnect?

JaseZa, 2020-03-31:
Hi Nuno, I saw that you added "Fixed MQTT connection on startup" to the changelog of 2.4.0.

I'm running 2.4.0 but still have connection issue. Stopping and starting MQTT fixes the problem. I have already built in a 180 second delay into starting SolPipLog, but this doesn't help.

I'm wondering if a simple solution would not be to try and reconnect every 10 seconds if the MQTT connection fails?

Log of failure and then restart and working:

_0
waterbessie
connect: false
server not responding or not found !
MQTT OFF
0
waterbessie
connect: true
ping request: true
MQTT ON
Connection Accepted_

JaseZa, 2020-03-31:
Okay, seems the issue is still on my side - can't get SolPipLog to start after Mosquitto Broker. But I think an attempt to reconnect may be a good idea anyway? Could be configurable?

JaseZa, 2020-03-31:
Managed to get it running consistently now - added a sleep for 45 seconds into the solpiplogstart.sh. Takes a long time for Hassio and Mosquitto broker to start!

njfaria, 2020-04-05:
Well good to know, i do try to make 5 attemps to reconnect. But it seems i need to wait some more time between the attemps. Thanks for reporting.

closed 2020-04-05 by njfaria