njfaria/SolPipLog #39: ANT BMS Support

efkostya, 2020-04-27:
Hi, sorry for my bad English. 
Thank you for the great program! I would be grateful, if you add support for ANT BMS (https://aliexpress.ru/item/32826820690.html), 
It is used by many people and it connects via Bluetooth to Raspberry without any problem
Protocol is simple, a virtual com port is used (/dev/rfcomm*) 
Script on Python available at (https://github.com/juamiso/ANT_BMS/blob/master/bms_post.py
I interested in monitoring parameters and switching SBU SUB depending on the SoC level inside Solpiplog. Thank you!

ojeysky, 2020-05-10:
I will like to add my +1 to this request, as am in the process of securing the BMS as well

njfaria, 2020-09-09:
looking at it...

lucasimons, 2021-09-02:
Any update of ANT BMS support?