njfaria/SolPipLog #40: MQTT not starting correctly on wifi disconnect with 2.4.0

PoloB12, 2020-04-28:
Hi, when my wifi disconnects, MQTT stops broadcasting. Activate and deactivate MQTT is enough to start it again. Any way to automate this ?

efkostya, 2020-04-28:
confirm. I have a similar problem

njfaria, 2020-05-02:
looking for solution... simulating it.

PoloB12, 2020-05-02:
By the way ; PVOutput continues normally and emoncms too, it's only MQTT that seems to be affected.

efkostya, 2020-05-03:
I temporarily solved the problem by installing mosquitto on the same raspberry and specifying localhost as the broker's mqtt address. 
Local Mosquitto configured the bridge mode to the server broker. This way it works fine. When I configure solpiplog to the server address directly-if the ping or connection is lost-mqtt itself is not restored, although sending data to pvonline does not stop.

PoloB12, 2020-05-03:
> I temporarily solved the problem by installing mosquitto on the same raspberry and specifying localhost as the broker's mqtt address.
> Local Mosquitto configured the bridge mode to the server broker. [...]

That's a smart solution, will look into that untill there's a final fix, thanks!

njfaria, 2020-05-03:
please try the new version... 
need feedback if it works with your configuration.

efkostya, 2020-05-03:
Thank you! I installed the new version, set it up on a remote mqtt  server, and wait.. still working

PoloB12, 2020-05-03:
Yes, it looks like the problem is solved after some quick testing with the 2.5 version. Thanks !

efkostya, 2020-05-04:
works fine thanks!

closed 2020-05-04 by njfaria