njfaria/SolPipLog #46: pylontech monitoring crashes

darkmordu, 2020-08-06:
When I connect to my pylontech pack, the values read seem corrupted
[image]
[image]

When value "number of packs" gets a crazy value, solpiplog crashes

darkmordu, 2020-08-06:
[image]

example of crash

njfaria, 2020-09-09:
HI, i am makeing a separate version for the pylontech, how many pylon packs do you got ?

darkmordu, 2020-09-10:
Hi, I have 4 batteries. It seems the time to read the batteries is too long. I tried it myself and it can take several seconds to get all info.

njfaria, 2020-09-11:
please try this version [pylog098.zip](https://github.com/njfaria/SolPipLog/files/5209673/pylog098.zip)
i am separating the pylon, batrium, bmv from solpiplog. more easy for debugging.
You need to deactivate the logging in solpiplog.
Extract file and execute.
Feel free to report bugs.

Jsilvasimoes, 2020-09-14:
Hello everybody, I Have the same problem with 2 pylon batteries and a easun SMVII 5k 48v MPPT.
I cannot get data from batteries using pylog098.

darkmordu, 2020-09-14:
I tried pylog098 but it does not work. Here is a screenshot
[image]

njfaria, 2020-09-14:
some buffer overflow. please try
[pylog099.zip](https://github.com/njfaria/SolPipLog/files/5220670/pylog099.zip)
any possibility to acces your raspberry ? You may send me a mail...

njfaria, 2020-09-14:
> Hello everybody, I Have the same problem with 2 pylon batteries and a easun SMVII 5k 48v MPPT.
> I cannot get data from batteries using pylog098.

Are you using a raspberry, please pay attention, software is working on raspberry only.
Are you using solpiplog ? If so, you need to stop the pylontech loggin in solpiplog.