njfaria/SolPipLog #60: Not possible to change battery charger mode via MQTT

Tomvleminckx, 2021-02-26:
Not possible to change battery charger mode via MQTT:

I've recently installed the version 3.0 to monitor my MPPsolar inverter.
All works great, really big fan of your app!
But I have an issue when trying to change the "Battery charger" modus via MQTT, on my MPPsolar inverter (5048GK also known as the Axpert-VM-III).
If I try the change the inverter's output priority via the topic solpiplog/pip/status/set with payload uti, sol, sbu this works.
But when I try to change the charger priority via solpiplog/pip/status/set with payload 0, 1, 2, 3 nothing is happening.
When I publish for example solpiplog/pip/status/set 1 and I check the status via debug (command QPIRI), then battery mode does not change.
According to the PIP30 protocol manual when the command QPIRI is issued Q should indicate the charger modus:
                                                                                                                          *
BBB.B CC.C DDD.D EE.E FF.F HHHH IIII   JJ.J KK.K JJ.J KK.K LL.L O PP QQ0 O P Q R SS T U VV.V W X
240.0 20.8 240.0    50.0 20.8 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 1 3 1 01 0 0 54.0 0 1    

As you can observe above the Q value is not changed.

Thanks,

Tom

Yuriyvp, 2021-04-01:
Same for me

njfaria, 2021-04-12:
sorry guys, the Readme is wrong it should be
solpiplog/pip/charger/set with payload 0, 1, 2, 3 for charger settings  

and sorry for the delayed reply.

Yuriyvp, 2021-04-13:
Hi.
I have tried with no success.
Anyhow thanks for your reply and for all your effort and work.

sijunair, 2021-05-29:
I use GODENERG along with solpiplog.
Solpiplog + GODENERG works well.

JaseZa, 2021-06-05:
Just confirming this isn't working for me either

JaseZa, 2021-06-17:
Figured this out - channel to write to should be:

solpiplog/pip/charge/set

closed 2021-08-18 by njfaria