njfaria/SolPipLog #61: MQTT QPIGS2 output

cybersyx, 2021-03-16:
Hi, this application is awesome. I send the data to an mqtt server without problems but as I could read from the documentation via mqtt it does not send the data of a second mppt (qpigs2). Is it possible to add this function? Thanks a lot !

njfaria, 2021-04-12:
need to add it

diybuc, 2021-05-09:
V3.0 seems to log my QPIGS2 once then it shows 0 for all values.
If i manually send QPIGS2 i get this reply
inverter model: 30
68 2D
neue version:QPIGS2h-
size : 9
02.8 248.7 00713

[image]
[image]
It is A V W
The only issue is under the main Windows it show QPIGS2 all 0 as values

It worked Flawless on the previous version of the Inverter but after a upgrade to the inverter it stopped.

njfaria, 2021-08-16:
will soon post the version which logs QPIGS2

njfaria, 2021-08-19:
@cybersyx @diybuc  version 3.1.1 online

closed 2021-08-19 by njfaria

diybuc, 2021-08-20:
You are the man. It is working a charm and I now realize how much Wh went
Mia due to my estimate calculation. I will make a small donation soon.
Thanks bud.