njfaria/SolPipLog #63: Configure MAX charging current (Total + UTI) via MQTT

myp-cz, 2021-04-19:
There are two items to write via MQTT - the mode (SBU/SUB/UTI) and submode (SBL/SLB UDC/UCD).
Which is great!

Could you add another two items to set via MQTT ?
That would help me to properly control the whole PV installation from a central point (RPi4).

MQTT: write "TOTAL CHARGING CURRENT" in Amps
MQTT: write "UTILITY CHARGING CURRENT" in Amps

This would directly translate to appropriate message to PIP-MK 5048 / Axpert King / etc.

JaseZa, 2021-06-05:
This would be a great feature!

njfaria, 2021-08-18:
@myp-cz and @JaseZa with **solpiplog/pip/maxcharge/set** and payloads **2**,**10**,**20**,**30**,**40**,**50**,**60** you may change the battery max charge current which is what you problably want.

JaseZa, 2021-08-18:
@njfaria this is great, thanks! Although it seems 50A and 60A doesn't work over MQTT? Just changes it to 40A instead. 50A works when I change it directly from Solpiplog though.

JaseZa, 2021-08-18:
Also just to note for others, this changes the "Max Utility Charge" not the Max total charge to the battery. In my case this is exactly what I wanted.

njfaria, 2021-08-18:
of course i uploaded the wrong version.. new one is online.

twmuller, 2021-11-04:
Hi, what a great piece of software. Thanks. My Max charge current can be set up to 140 Amps on the invertor and i want to be able to set it via MQTT, but reading the info, it looks as if it can only go to 60Amps via the software. True or can i send 140 via MQTT?

Thanks