njfaria/SolPipLog #65: Victrom BMV

Zippa1204, 2021-08-06:
Hi,

I have a Victron BMV702. Does it also work?

Am I missing something how to connect, as I do not see the USB port in the selection?

Gert

Ricknanon, 2021-08-08:
Hello. 
You have to install Victron battery monitor (from same repo) and this app you will find the usb to connect. You will need as well to enable mqtt if you want the threshold to work with solpiplog ( ie at 70% of discharge go back to grid in solpiplog)
Have a good day.

Zippa1204, 2021-08-08:
Hi,

Yes, I have the APP, but when I want to connect, the USB port listed seems odd - or is it correct

Gert

Ricknanon, 2021-08-08:
Yes it is those odd names. Try to connect any the app will tell you if it is ok.

Zippa1204, 2021-08-16:
Hi, OK I opted for Daly BMS.
However, see the cell voltages.....this is creating bad graphs on Emoncms.
[image]

Ricknanon, 2021-08-16:
It is not supposed to work with Daly if you have a BMV !

njfaria, 2021-08-16:
@Zippa1204 if you got a BMV you should use https://github.com/njfaria/bmvlog

Zippa1204, 2021-08-17:
Hi,

Yes, aware of that. Justto clarify:
My Friend has Victron BMV.
I have Daly BMS.

I have shared the pic from my Daly BMS, using the DalyLog App.

I also noted that the SOC switching is not happening when I selected it. I set to 20%, but batteries went down to 15%, when inverter switched back to grid.

Gert

Ricknanon, 2021-08-17:
Ok 👍 
I ll try on my side for soc. 
Did you check on solpiplog tha soc is ok when running bmvlog. 
They need to run both and to be mppt connected. 
Rick

Zippa1204, 2021-08-17:
Hi,

Yes, both runnig on same Raspberry, as well as MQTT running on both apps. On SolPipLog, I can see the SOC value passed from DalyLog - so that seems to work correct.

Gert

Ricknanon, 2021-08-17:
I will change the value on mine to check.

Ricknanon, 2021-08-18:
Hello i tried on my side and it is working fine.  
« Soc is low «  is depicted on solpiplog switch tab. 
And my inverter switched to utility.

Zippa1204, 2021-08-19:
Hi,

OK, Maybe I am doing something wrong then, as mine never shows SOC is low.

I saw that you do not have access to Daly BMS on Pi etc - I am willing to give you VNC access to my Pi should you wish?

Gert

njfaria, 2021-08-19:
just invite solpiplog at gmail dot com to your vnc

à(s) 17:43: