njfaria/SolPipLog #73: QPIGS2 is not sent via MQTT

ArminK1337, 2022-02-07:
QPIGS is sent over MQTT and QPIGS2 is not.

ArminK1337, 2022-02-13:
QPIGS2 it´s showing on SolPipLog but not send over MQTT

ArminK1337, 2022-02-13:
[image]
[image]

ArminK1337, 2022-02-13:
ja dann kann man sich auch auf deutsch unterhalten :D

ArminK1337, 2022-02-13:
Perfekt Daten werden jetzt übertragen :D Vielen Vielen Dank und mega Software, bin sehr zufrieden :D

ArminK1337, 2022-02-13:
Okay ich hab nen Bug gefunden bei der  3.2.0 ist ja auch ne "Beta" Sobald es die Ladequellen Priorität umstellen will von "Solar" auf "Solar and Utility" stürzt die ganze Software ab.

njfaria, 2022-02-15:
womit ? per MQTT oder interface ? komplett absturz ist natürlich extrem...

ArminK1337, 2022-02-15:
Bei beidem, spielt keine Rolle

njfaria, 2022-02-15:
mail mich mal an solpiplog at gmail.com

closed 2023-08-06 by njfaria