njfaria/SolPipLog #81: problem with infinisolar sp24 twin

Maxym77, 2022-10-26:
Hi, 
when i click "connect" in /dev hidraw0
there is no data from inverter.... you can help me?

Oct 26 17:06:20 energiadomestica mtp-probe: bus: 1, device: 35 was not an MTP device
Oct 26 17:06:21 energiadomestica kernel: [  707.280771] usb 1-1.1: USB disconnect, device number 35
Oct 26 17:06:23 energiadomestica kernel: [  709.152942] usb 1-1.1: new full-speed USB device number 36 using xhci_hcd
Oct 26 17:06:23 energiadomestica kernel: [  709.295113] usb 1-1.1: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
Oct 26 17:06:23 energiadomestica kernel: [  709.295136] usb 1-1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=0
Oct 26 17:06:23 energiadomestica kernel: [  709.295151] usb 1-1.1: Product: HID in FS Mode
Oct 26 17:06:23 energiadomestica kernel: [  709.295162] usb 1-1.1: Manufacturer: STMicroelectronics
Oct 26 17:06:23 energiadomestica kernel: [  709.309329] hid-generic 0003:0665:5161.0022: hiddev96,hidraw0: USB HID v1.11 Device [STMicroelectronics HID in FS Mode] on usb-0000:01:00.0-1.1/input0
Oct 26 17:06:23 energiadomestica mtp-probe: checking bus 1, device 36: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Oct 26 17:06:23 energiadomestica mtp-probe: bus: 1, device: 36 was not an MTP device
Oct 26 17:06:23 energiadomestica mtp-probe: checking bus 1, device 36: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"

lucasimons, 2023-01-10:
@Maxym77 Hai risolto? Sei riuscito a leggere qualcosa?

Maxym77, 2023-01-10:
> @Maxym77 Hai risolto? Sei riuscito a leggere qualcosa?

si ho risolto.
E' bastato NON utilizzare il cavo USB e usare la seriale normale.

Va una meraviglia.

Via usb i pacchetti vanno tagliati a 8 byte mentre con la seriale no.

Evidentemente il calcolo crc con i tagli a 8 byte e la virgola fanno qualche casino.. 
Non tagliando a 8 byte tutto si è risolto, quindi seriale e via.

lucasimons, 2023-01-10:
> Non uso solpiplog. Uso altri tool.

cosa usi? hai per caso provato pipsolar con esphome?

Maxym77, 2023-01-10:
uso iverter-tools

antonioste, 2023-02-19:
a quanto sembra ci sono alcuni inverter che non usano il protocollo pip, tra cui i nuovi easun...
non sono ancora riuscito a trovare un tool semplice che mandi i dati su emoncms... 
Qualcuno ha notizie in merito?