njfaria/SolPipLog #89: Solpiplog with three inverters in three-phase configuration

Willy76898, 2023-07-08:
I have a system with three inverters in a three-phase configuration (one phase inverter). I installed solpiplog and by connecting the three usb cables to the raspberry I can only see one inverter, the one I choose in the initial configuration (hidraw0 port). Am I doing something wrong or is it really not possible to monitor the three inverters at the same time?

njfaria, 2023-08-06:
The normal situation is that the parallel interface is installed, so you should use parallel in solpiplog and since they are in parallel only one cable is needed. If that is not the case you need to create separate folders sol1, sol2, sol3 for example. Copy solpiplog to that folder and execute from there with the correct interface. hidraw0 and so on.

Mattiasimon, 2023-09-17:
Good evening, my MUST POWER inverter is compatible with the Solpiplog program? if I connect the serial port of my inverter with a port on the Raaspberri with a cable, it doesn't detect the inverter values.