njfaria/SolPipLog #97: Charging and Bypass mode

fox25580, 2024-01-07:
HI Nuno,
your app it is wonderfull!!
i have 2 questions:

it is possible to start and stop the charging by MQTT?
it is possible to move to BYPASS mode by MQTT?

njfaria, 2024-02-10:
[https://github.com/njfaria/SolPipLog/blob/master/docs/mqttvalues.md](https://github.com/njfaria/SolPipLog/blob/master/docs/mqttvalues.md)
change to BYPASS you probably are mentioning changing to solar or grid ? 
_solpiplog/pip/status/set uti_ for example changes to grid
payloads are 
**uti** (grid )
**sbu** ( solar->battery->grid )
**sol** ( only solar )
some inverter do have **aut** **onl** and **eco** as payloads
as for charging :
_solpiplog/pip/charge/set 0_ changes to charge from grid
payloads
**0** grid first 
**1** solar first ( no solar, starts to charge from grid )
**2** solar and grid ( not all inverters support this mode )
**3** solar only

kxn25, 2024-03-13:
solpiplog/pip/status/set uti for example changes to grid

How can i change fom Line mode in Battery mode and back?
I have a Effekta AX-M2 3000.